mariolukas/Bluetti_ESP32_Bridge #2: Generally runs with the EB3A

tomschlde, 2022-11-16:
The connect works, but the data points are generated only sometimes, then also not always completely.
But you can also create them manually (the entries in the log reveal the names). Then the values are updated correctly in the IOBroker.

[image]

Good work so far - thanks.

Regards
Thomas

giovanne123, 2023-01-18:
Hi Thomas,
in the specific DEVICE_EB3A.h (https://github.com/mariolukas/Bluetti_ESP32_Bridge/blob/main/Bluetti_ESP32/DEVICE_EB3A.h) have you left it at the beginning with:
```
#ifndef DEVICE_AC300_H
#define DEVICE_AC300_H
```
or changed to?:
```
#ifndef DEVICE_EB3A_H
#define DEVICE_EB3A_H
```

Have you changed something to get it running with EB3A?
I ask because I have some trouble get it running. Maybe you can have a look? https://github.com/mariolukas/Bluetti_ESP32_Bridge/issues/20

mariolukas, 2023-01-21:
This is fixed in code now.

closed 2023-01-21 by mariolukas