mariolukas/Bluetti_ESP32_Bridge #6: Compilation error: 'class BLEClient' has no member named 'setMTU'

mbay0r, 2022-12-10:
hi, i cant compile the code.
i would like to use it with an ESP32 board with CP2102 for my AC200MAX.

I always get the following error:

`Bluetti_ESP32\BTooth.cpp: In function 'bool connectToServer()':
Bluetti_ESP32\BTooth.cpp:117:14: error: 'class BLEClient' has no member named 'setMTU'
     pClient->setMTU(517); //set client to request maximum MTU from server (default is 23 otherwise)
              ^

exit status 1

Compilation error: 'class BLEClient' has no member named 'setMTU'`

thank you.

mbay0r, 2022-12-11:
did use an old esp32 libary. :-(

nygma2004, 2023-01-19:
I am getting the same message. I updated my ESP32 board and it did not help. Also checked my libraries, but I am not seeing any Bluetooth library. Where do you update this?

mariolukas, 2023-01-21:
i can confirm that. Same error trying to build the current master branch with platformIO

mariolukas, 2023-01-21:
fixed it by updating my platform libs

`pio pkg update`

closed 2023-01-21 by mariolukas

nygma2004, 2023-01-21:
Anyone have idea how to make this work in Arduino IDE? I downloaded the new project, but I am still getting the error. I updated my boards in board manager. In the library manager, I don't see any bluetooth, therefore it must be part of board package.

gerab, 2023-01-29:
@nygma2004 
answered here:
https://github.com/mariolukas/Bluetti_ESP32_Bridge/issues/9#issuecomment-1351685553

nygma2004, 2023-01-29:
@gerab Thanks a lot. I never throught that my library references are outdate. It is all working now.