mariolukas/Bluetti_ESP32_Bridge #8: works good on ac200max, but please implement more

mbay0r, 2022-12-11:
it works very well with ac200max, thanks for the great work!

but unfortunately not much is supported yet, maybe you could add the rest of https://github.com/warhammerkid/bluetti_mqtt/blob/main/bluetti_mqtt/core/devices/ac200m.py ?

thanks a lot

viper4gh, 2022-12-23:
Hi,
I used it with an M5Stack Atom Lite and it works good. The Atom has a built-in RGB LED and I added some lines for activating different colors for different connect states like WiFi, MQTT, Bluetooth connected, which helps with the following.
I have sometimes connect issues on WiFi or Bluetooth after multiple resets, but I open a seperate issue for it.

If it is possible I have the same request for adding the rest of the values which are provided by the python script.
Further it would be nice if the ac_power_on and dc_power_on are command datapoints also available as state, because with the command datapoints you don't know the current state at the powerstation.

AlexBurghardt, 2023-01-20:
I have improved several things. Now I have 20+ values available in iobroker (via mqtt) including the ac/dc states of my AC200M.
Soon you will see it on here github..

DraycD, 2023-01-21:
> I have improved several things. Now I have 20+ values available in iobroker (via mqtt) including the ac/dc states of my AC200M.
> Soon you will see it here on github..

Will they also been shown on the EP500P ?

mariolukas, 2023-01-21:
> Will they also been shown on the EP500P ?

When the address range of the EP500P is the same as for the already supported devices it should work. Otherwise we need to reverse engineer on a EP500P.

mariolukas, 2023-01-21:
> I have improved several things. Now I have 20+ values available in iobroker (via mqtt) including the ac/dc states of my AC200M.
> Soon you will see it here on github..

I would appreciate a PR.

AlexBurghardt, 2023-01-30:
@mariolukas: PR is done, please check.

Since this is my very first contribution, I will also have to learn how Git/Github works in detail.
Today I synchronized my fork with your main and that led to a second commit in my PR. Don't know why. The first commit contains my improved code.

mariolukas, 2023-01-30:
Yes, thank you for your commitment. I appreciate it a lot.

But it will take me some time to merge this into actual code. 

>  I will also have to learn how Git/Github works in detail.
One suggestion for your next PR. Just do your changes on the current code in the code folder and do a PR. That makes merging a lot easier for me. 

I need to compare each single file of your PR #33 in the main folder with the current code base in my source folder and cherry pick the changes you've made.

AlexBurghardt, 2023-01-30:
Yes, no diffs are displayed. I noticed it and I try to do it the way it should be done next time.
Feel free to ask me about my changes. Also a Webex live review would be possible to make it easier for you to understand the changes.

AlexBurghardt, 2023-02-11:
@mariolukas: Would it be better to close the pull request, updating my repository to the current status/code and committing my changes to it? Then I could make specific PRs with differences being displayed. But that would take a few days more...

mariolukas, 2023-02-12:
Yes i think that would be a better way to merge the PR. Please leave a comment to the PR and describe what you changed ( no details needed, only what you fixed or which new features are added ...). Thanks!

cybercreature, 2023-02-17:
> it works very well with ac200max, thanks for the great work!
> 
> but unfortunately not much is supported yet, maybe you could add the rest of https://github.com/warhammerkid/bluetti_mqtt/blob/main/bluetti_mqtt/core/devices/ac200m.py ?
> 
> thanks a lot

Hello,

how did you manage the connection to the Bluetti. I cannot connect to the AC200MAX.

Did you have to make any changes to the code?

thanks