mariolukas/Bluetti_ESP32_Bridge #9: During compiling I got BLEDevice.h No such File or directory

BS-GIFHUB, 2022-12-13:
Any help on this issue ?

Thanks

Bernhard

PS: Sorry I'm new with ESP32 programming

mariolukas, 2022-12-13:
It seems that the bluetooth library is missing. Have you installed the correct board support package for your ESP32. This lib is shipped by the board support package as far as i know.

BS-GIFHUB, 2022-12-14:
Okay thanks. I did a reinstallation of the Ardiuno with ESP32 and now I got the following issue

C:\Users\besa\Documents\Arduino\sketch\Bluetti_ESP32\BTooth.cpp:116:14: error: 'class BLEClient' has no member named 'setMTU'
     pClient->setMTU(517); //set client to request maximum MTU from server (default is 23 otherwise)
              ^

exit status 1

Compilation error: 'class BLEClient' has no member named 'setMTU'

mbay0r, 2022-12-14:
you are using an old esp32 libary.

Go to File -> Preferences -> Additional Boards Manager Urls  and add
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

then install in the board manager the newest esp32 version 2.0.5.

BS-GIFHUB, 2022-12-14:
Thanks for the quick and magical help which solves the problem. 

Stay Healthy. Bernhard.

closed 2022-12-14 by BS-GIFHUB