mariolukas/Bluetti_ESP32_Bridge #10: No BT connection

BS-GIFHUB, 2022-12-14:
I own a AC500 and with bluetti-mqtt I got very quick a connection. 

Now using the ESP32 I'm not getting an connection to my Bluetti station via BT. device id was checked 10 times and is fine. Connection to the mqtt server is fine but no connection the the Bluetti.

Any ideas ?

Thanks Bernhard

BS-GIFHUB, 2022-12-15:
I can test a connection with a manual setting 

of bluettiDevice = "B8:D6:1A:BF:32:EB"

but I don't the syntax how to overwrite 
bluettiDevice = new BLEAdvertisedDevice(advertisedDevice);

so that I'm able to use a fix id (not from you scan routine) which i got from "bluetti-mqtt --scan" using my PI.

So if you help I can test the issue with my AC500. 

Greetings Bernhard

BS-GIFHUB, 2022-12-15:
Okay, 

now I installed the BLE Scanner from https://github.com/topics/ble-scanner to test my ESP32 Bluetooth and all is fine: BLUETTI was found and listed.

So now my conclusion is that something may be wrong with the used Bluetooth Library or routines because only few BT devices were listed and the signal from the BLUETTI isn't very strong (tested with a Bluetooth Scanner on my Iphone). So it doesn't connect. 

Any ideas to improve the Bluetooth performance in this module ?

Thanks 

Bernhard

BS-GIFHUB, 2022-12-16:
Changed in BTooth.cpp

The if condition to 

if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) ) {

Now I have the connection with the AC 500, but.......

no AC DC power status and command

MartinRGH, 2022-12-29:
Hello Bernhard,

I have changed the line to

`    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) && (strcmp(advertisedDevice.getName().c_str(),settings.bluetti_device_id)==0) )
`
otherwise the device name is not checked at all and it will not be possible any more to have more than one devices.

Greetings,
Martin

mariolukas, 2023-01-21:
Merged some of the PR's those should fix connection issues.

closed 2023-01-21 by mariolukas