mariolukas/Bluetti_ESP32_Bridge #15: WiFi sometimes not connecting after reset

viper4gh, 2022-12-23:
During experimenting with a M5Stack Atom Lite and the code I did several resets and sometimes the device was not able to connect to WiFi and activated the local Access Point instead.
I don't know how the connect is implemented exactly, but it seems that it is trying the connect only one time and if it cannot connect it switches to the local AP.
Is it possible to implement that the device trying to connect to the configured WiFi two or three times with an interval of for example 2 seconds in between, before it switches to the local AP.
Thanks in advance.

areyoufserious, 2022-12-29:
I have worse case
esp can't connect to wifi at all, after providing wifi cridintials it reboots but with own AP
at the /info page i see "Not connected to {name of my WiFi SSID}"
looks like it never have been connected to WiFi
How to fix this?

mariolukas, 2023-01-21:
I just merged some PR's this should fix some of the connection issues with BT and Wifi.

closed 2023-02-21 by mariolukas