mariolukas/Bluetti_ESP32_Bridge #17: No state dc/ac_output_on

motoolli, 2023-01-09:
Contrary to the description, the status of dc/ac_output_on cannot be queried. 
MQTT Explorer does not show me any values. ( I use Bluetti AC200M )
----------------
state
device_type = AC200M
serial_number = xxxxxxxxxxxxxxx
arm_version = 4008.03
dsp_version = 4005.07
dc_input_power = 0
ac_input_power = 0
ac_output_power = 0
dc_output_power = 0
power_generation = 0.00
total_battery_percent = 62
device = {"IP": "10.10.10.xx", "MAC": "AA:BB:CC:DD:EE", "Uptime":37857652}
----------------

giovanne123, 2023-01-19:
Same on EB3A for me.
Also for some other details compared to original bluetti_mqtt (see https://github.com/mariolukas/Bluetti_ESP32_Bridge/issues/20#issuecomment-1396635734)
I think there is some more device specific work needed (the device files seems not customized so far)... will look if I find some progress for EB3A with my limited knowledge ;-) ...

But for the 'dc/ac_output_on' I also have no idea so far :-(

AlexBurghardt, 2023-01-20:
I made various improvements. This issue I fixed at first. I know what is going wrong. There is some code in Payloadparser.cpp which causes the Problem. Since I am new to github I asked mariolukas if I shall upload my whole project as a pull request.
Once I get an answer you will see what I have done to fix it. Now I get more than 20 values from my AC200M to my iobroker.

giovanne123, 2023-01-21:
> 

Can you post the changes in Payloadparser.cpp here, than we can test it and I can also check if it is working for EB3A too?

AlexBurghardt, 2023-01-29:
Pull request is done. You should be able to see my code.

giovanne123, 2023-01-29:
@AlexBurghardt, thanks I will give it a try...

motoolli, 2023-01-30:
Hello,
after trying everything possible to compile and upload, I can't get a different result than the current version 0.1.0

last messages (time / value):
642432: device_type -> AC200M
642436: serial_number -> 2xxxxxxxxxx
642438: arm_version -> 4008.03
642441: dsp_version -> 4005.07
642442: dc_input_power -> 18
642454: ac_input_power -> 0
642456: ac_output_power -> 0
642458: dc_output_power -> 0
642459: power_generation -> 0.00
642461: total_battery_percent -> 16
723483: device_type -> AC200M

I am not very versed in programming etc, can someone help me? What is confusing me is the nested file structure after resolving the ZIP file from AlexBurghardt.

Thank you Oliver

Translated with www.DeepL.com/Translator (free version)

AlexBurghardt, 2023-01-30:
Well, if you download from my fork you need to merge the files  *.ino, *.h and *.cpp from root diectory of the zip file into the <Bluetti_ESp32> directory that is in the zip file.
In other words: the root directory of the zip file contains my improvements and the files inside of the <Bluetti_ESp32> directory must be overwritten by them. Afterwards you can use all files from the <Bluetti_ESp32> directory to build the firmware.

Attention: you will get a compiler error message if you don't install the  the U8g2 library (2.33.15).

motoolli, 2023-01-30:
@AlexBurghardt , WOW, you made my day. The MQTT-Explorer shows me a lot of values now. 
Thanks a lot, now i will test the dc/ac_output_on.

AlexBurghardt, 2023-01-30:
Your welcome, enjoy it. If you connect a 128x64 OLED you will see the WiFi and BT connection status of the ESP on it.

mariolukas, 2023-01-30:
I will merge @AlexBurghardt's code until the end of the week.

closed 2023-02-21 by mariolukas