mariolukas/Bluetti_ESP32_Bridge #19: no bluetooth connection AC500

ingozilla, 2023-01-18:
Hello Mario,
many thanks for your super project!!!!
 
I have an AC500.
I installed my own HomeAssistant server with Mosquitto MQTT :-)
with "bluett-mqtt" everything works fine too.
 
with ESP32 I am still at war.
When compiling pclient MTU was criticized, I simply commented out the line,
after that I could compile and start the ESP32.
 
I have also included the changes mentioned below, but I can't get a bluetooth connection to the AC500,
do you have any idea what I am doing wrong?
 
Many greetings
Ingo

config.h
 
#ifndef BLUETTI_TYPE
#define BLUETTI_TYPE BLUETTI_AC500
#endif
  
//pBLEScan->setInterval(1349);
//pBLEScan->setWindow(449);
//pBLEScan->setActiveScan(true);
//pBLEScan->start(5, false);
pBLEScan->setInterval(1349);
pBLEScan->setWindow(5000);
pBLEScan->setActiveScan(true);
pBLEScan->start(40, false);
 
 
BTooth.cpp
 
if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) && (strcmp(advertisedDevice.getName().c_str(),settings.bluetti_device_id)==0) ) {

mariolukas, 2023-01-19:
> many thanks for your super project!!!!
Thank you. 

I think i need to merge the open PR's.

mariolukas, 2023-01-21:
Are you using Arduino or PlatformIO for building?

> When compiling pclient MTU was criticized, I simply commented out the line,
> after that I could compile and start the ESP32.

This is caused by an old Board Support Package for the EPS32. Can be easy fixed by updating all the EPS32 stuff. If you are using PlatformIO you should just run 

`pio pkg update`

All open PR's are merged now. Maybe this will fix your issues.