mariolukas/Bluetti_ESP32_Bridge #20: Trying to get it running for EB3A

giovanne123, 2023-01-18:
Hi,
I'm trying to get it running for a EB3A.

- First question, is it right that the specific DEVICE_EB3A.h (https://github.com/mariolukas/Bluetti_ESP32_Bridge/blob/main/Bluetti_ESP32/DEVICE_EB3A.h) file is starting with:
#ifndef DEVICE_**AC300_H**
#define DEVICE_**AC300_H**

- I changed it to:
#ifndef DEVICE_EB3A_H
#define DEVICE_EB3A_H

- But at the moment I only get Wifi connected and Mqtt-Broker connected.

**But BL is not connecting:**
The Bluetti Bluetooth ID is the "EB3A2249........", right?
```
C:\Users\Max>bluetti-mqtt --scan
Scanning....
Found EB3A22490xxxxxxxxxx: address DD:E8:6C:xx:xx:xx
```

WebUI:
[image]
Serial-Output (some text replaced with XXXXXX):

```
*wm:STA IP Address: 192.168.0.13

IP address: 
192.168.0.13
MDNS responder started
HTTP server started
BLE Advertised Device found: Name: EB3A2249XXXXXXXXXX, Address: dd:e8:6c:xx:xx:xx, manufacturer data: 424c424c554xxxxxx, serviceUUID: 0000ff00-0000-1000-8000-00805f9b34fb, rssi: -64
...
rssi: -75, serviceData: ���a���¢-�r
>89
Connecting to MQTT at: 192.168.0.111:1883
Connected to MQTT Server... 
```

Mqtt:
[image]

**- Is there someone already using it with EB3A?
Have you done changes?**

giovanne123, 2023-01-18:
Hm, flashed the fork: https://github.com/Seb0815/Bluetti_ESP32_Bridge
And this is directly working with EB3A
@Seb0815, Home Assistant Auto Discovery would be fine ;-)

[image]

[image]

giovanne123, 2023-01-19:
Update:
Connect to the EB3A with original from Mario is also working, when modifying the
BTooth.cpp:
Line 50:
from:
` if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) && advertisedDevice.getName().compare(settings.bluetti_device_id)) {`
to:
 `if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) && (strcmp(advertisedDevice.getName().c_str(),settings.bluetti_device_id)==0) ) {`

[image]

@mariolukas, can you please update this to support these devices? (maybe test before if still working for your device)
@Seb0815, can you also update this in your fork? Maybe I will use this because of the nice WebUI ;-)
 
Now, I think details are missing in MQTT, see pictures for comparison with bluetti_mqtt:
- ac_output_on
- dc_output_on
- led_mode
- power_off
- eco_on
- eco_shutdown
- charging_mode
- power_lifting_on
- (need to test the possible commands)

additional nice features:
- (wifi retry if not connecting at boot)
- (HA discovery)
- (publish intervall changable by WebUI / command)
- (...)

ESP32-Bridge:
[image]

bluetti_mqtt (python):
[image]

mariolukas, 2023-01-19:
I will merge all open PR's this week, this will bring a new Webserver interface. Additionally i will fix the device definition in the header for the EB3A.

giovanne123, 2023-01-19:
> I will merge all open PR's this week, this will bring a new Webserver interface. Additionally i will fix the device definition in the header for the EB3A.

Can you please also modify the 
BTooth.cpp: line 50
See above.

mariolukas, 2023-01-19:
Yes i will review this.

Seb0815, 2023-01-19:
@giovanne123 WLAN reconnect is already in my latest fork. I will create a PR as soon as @mariolukas merged all other PR's

giovanne123, 2023-01-19:
> Yes i will review this.

 👍

giovanne123, 2023-01-19:
> @giovanne123 WLAN reconnect is already in my latest fork. I will create a PR as soon as @mariolukas merged all other PR's

 👍

areyoufserious, 2023-01-20:
i edited BTooth.cpp and DEVICE_EB3A_h as suggested
but stiil no connectiion

@giovanne123 asked
> The Bluetti Bluetooth ID is the "EB3A2249........", right?

i also would like to know

giovanne123, 2023-01-20:
> i edited BTooth.cpp and DEVICE_EB3A_h as suggested but stiil no connectiion
> 
> @giovanne123 asked
> 
> > The Bluetti Bluetooth ID is the "EB3A2249........", right?
> 
> i also would like to know

Starting with the EB3A... would be right. With this it is connecting for me (see above screengrabs)

mariolukas, 2023-01-21:
I am closing this because the initial issues should be fixed now. Reopen it if you have further issues related to this topic, otherwise create a new issue.

thanks !

closed 2023-01-21 by mariolukas

giovanne123, 2023-01-21:
Is ok. I will/have create/created new issues.
Was more collection for me what I will have to look for ;-)

giovanne123, 2023-01-21:
> @giovanne123 WLAN reconnect is already in my latest fork. I will create a PR as soon as @mariolukas merged all other PR's

@Seb0815, I think you can create your PR regards to WLAN retry if not connected after first boot, because it isn't working [https://github.com/mariolukas/Bluetti_ESP32_Bridge/issues/26] after Mario has merged the PRs.