mariolukas/Bluetti_ESP32_Bridge #22: [Question] Is connection between Bluetti and ESP32 stable for you?

giovanne123, 2023-01-19:
Is the connection between Bluetti and ESP32 stable for you, when running long time?

At the moment I think after some time the Bluetooth in the EB3A is turned Off?!

After some time I see that the ESP32 is longer sending and in serial terminal I see that the ESP32 does no longer see the Bluetti.
After pushing a power button on Bluetti, ESP32 continues working and sending data.

EB3A not powered from the Bluetti itself.
(Maybe ESP32 powered by Bluetti keeps Bluetooth online? Not tested so far)

EB3A EU - firmware:
ARM 2053.07
DSP 2052.07
Powered not from Bluetti 

I am interested in behaviour for you and especially with EB3A and used firmware versions? And how it is powered?

Seb0815, 2023-01-19:
The stability of the connection was my motivation to add some of the code. My Bluetti (EP500P) will serve some of my critical devices (servers, internet, network .. all the smarthome which control heating and so on) with power and especially in winter I need to make sure I know the power state of the battery to enable gird if necessary.
At the beginning of my journey I lost connection regularly....with the current code of my Bluetti_ESP32_Bridge it's stable since 3 weeks. The ESP reboot every now and then but it's back in seconds. In the case of emergency (device does not come back) the relays mode triggers an Shelly to enable grid mode. 
However sometimes the Bluetti send strange BT data via MQTT and the only way to fix this is to turn the EP500P off/on.

giovanne123, 2023-01-20:
> The stability of the connection was my motivation to add some of the code. My Bluetti (EP500P) will serve some of my critical devices (servers, internet, network .. all the smarthome which control [...]

How is your ESP powered? (by Bluetti, by external Power Plug, ...)

Seb0815, 2023-01-20:
> 
external to prevent a single point of failure

giovanne123, 2023-01-20:
> > 
> 
> external to prevent a single point of failure
👍

AlexBurghardt, 2023-01-20:
I think, the less the distance between ESP32 and the Bluetti the more stable is the connection. I request one big BT telegram (254 byte) from the Bluetti and it works best if the ESP32 is close to the powerstation.

giovanne123, 2023-01-21:
I think important is also the Bluetti device type.
For me EB3A (EU Version, important because of different voltage and firmware versions delivered)

I made the test today, to have the EB3A (with nothing attached) connected to the official Bluetti app on my android device. I hold the app/display on.
After round about 1h the Bluetooth connection in the App was lost :-(

I have to check how to let the Bluetooth stay connected (maybe trigger an action on the EB3A which hold Bluetooth online, ...)
I know Bluetooth is draining battery but unfortunatelly so far I can only switch it on again by pressing one of the power buttons.

Maybe there can be some improvements by FW updates.
So far there are only newer FWs available for NOT EU models of EB3A (adressing lots as I can read in the internet) :-(

mariolukas, 2023-01-21:
All PR's are merged now.

mariolukas, 2023-01-21:
My AC300 changes into a power save sleep mode when it is not used, even if BT is activated. Maybe this can cause also connection issues.

giovanne123, 2023-01-21:
Ok, close it because reason is not in ESP32 solution. In ESP32 I can only see later some possible workaround for holding the bluetooth alive (e.g. by sending an action via mqtt to the device).
But for the moment it is more related to the device itself. 
Have contacted Bluetti for the question regarding bluetooth behaviour and possible solutions.

closed 2023-01-21 by giovanne123

gerab, 2023-01-29:
Hey @giovanne123, any updates from Bluetti?

giovanne123, 2023-01-29:
Yes, but only default answer that they will improve continously, but nothing concrete so far. (In != EU they already released newer firmwares :-( , but for other issues as readable in Internet/forums)

For bluetooth on EB3A I decided to go a different way and only switch the EB3A on when needed or switch in high intervals on to not consume so much power from the Bluetti (Servo attached to ESP32 #30) because it is the little one and continiously bluethooth on is not that good.
But still investigating... ;-)