mariolukas/Bluetti_ESP32_Bridge #24: Lots of improvements made, what shall I do?

AlexBurghardt, 2023-01-20:
Hi Mariolukas,

first of all a big "thank you"  to you and all contributors. I got my AC200M in December and soon after I found your project, based on the Python project.

I got my first ESP32 NodeMCU and startet to work with C++ for the first time.

Long story short: I took your source code and implemented the changes  from the other contributors and started to understand how it all works. In the beginning I got only 10 values into my iobroker but now it works very well and I implemented more and more.
For instance I see the voltage of each single cell in my iobroker.
The reason for not seeing AC/AC on/off via mqtt was an issue with the PayloadParser. I reworked the "field in range" filter completely.
I also added 128x64 OLED support. There are a lot of changes I made.

There are still open points with my code: Since all code is executed in the standard loop() the display is hanging when ,for instance, 
the BT connection dropped.

I tried to create a new task and put the display handling in there, but that caused other problems related to the processors workload.

Shall I store my version as a pull request?

It is a mixture of your source, some other pull requests and my improvements.
This also will be my first contribution on github.

Best,
Alex

mariolukas, 2023-01-21:
Hi Alex, 

Thank you! I just merged all the open PR's. I would appreciate a PR. I think there will be a lot of conflicts but i think we can resolve all this. Just do a PR and we will find out.

- Mario

giovanne123, 2023-01-21:
> Hi Mariolukas,
> ...
> Long story short: I took your source code and implemented the changes from the other contributors and started to understand how it all works. In the beginning I got only 10 values into my iobroker but now it works very well and I implemented more and more. For instance I see the voltage of each single cell in my iobroker. The reason for not seeing AC/AC on/off via mqtt was an issue with the PayloadParser. I reworked the "field in range" filter completely. I also added 128x64 OLED support. There are a lot of changes I made.
> ....

Would be fine to see your changes, especially for the more values and also the ac/dc_on/off for the mqtt would be fine.

AlexBurghardt, 2023-01-21:
Hi Mario,

Just let me do some "cleanup" and then I will upload all files.
I'm a friend of commenting.
I did this on my changes and additional code but also added some comments to code that I did not change.
I hope, this is ok for you.

Alex

AlexBurghardt, 2023-01-28:
Please be patient. I will upload my code ASAP. Been busy with other things, sorry for that.

closed 2023-02-21 by mariolukas