mariolukas/Bluetti_ESP32_Bridge #25: still no bluetooth connection for EB3A

giovanne123, 2023-01-21:
Compiled lastest code from today, still no bluetooth connection for EB3A without:

BTooth.cpp:
Line 50:
from:
` if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) && advertisedDevice.getName().compare(settings.bluetti_device_id)) {
`to:
`if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) && (strcmp(advertisedDevice.getName().c_str(),settings.bluetti_device_id)==0) ) {
`
Serial.out:
```
192.168.0.13
MDNS responder started
HTTP server started
BLE Advertised Device found: Name: EB3A224xxxxxxxxxxxxx, Address: dd:aa:bb:cc:97:c9, manufacturer data: 424c424c5545545449, serviceUUID: 0000ff00-0000-1000-8000-00805f9b34fb, rssi: -50
...
Connecting to MQTT at: 192.168.0.111:1883
Connected to MQTT Server... 
subscribe to topic: ac_output_on
subscribe to topic: dc_output_on
Client reconnected! Last message ID that it got is: 57002

```
platformio.ini
+ ESP32 BLE Arduino @ 2.0.0
+ monitor_speed = 115200

PS C:\Users\Bluetti_ESP32_Bridge> pio pkg update
Resolving esp32dev dependencies...
Already up-to-date.

giovanne123, 2023-01-25:
No further investigation needed, I will add the change to my special usecase for the little EB3A.

closed 2023-01-25 by giovanne123

gerab, 2023-01-26:
@giovanne123 
`advertisedDevice.getName().compare(settings.bluetti_device_id) `

doesn't work for me either.  It's not only your device which has problem connecting.