mariolukas/Bluetti_ESP32_Bridge #26: still no retry if first Wifi isn't connected after boot

giovanne123, 2023-01-21:
Compiled lastest code from today, still no retry if first Wifi isn't connected after boot:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
deactivate relais contact
Loading Values from EEPROM
*wm:AutoConnect
*wm:Connecting to SAVED AP: MY_WLAN
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: FAILED 
*wm:StartAP with SSID:  Bluetti_ESP32
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal
ets Jun  8 2016 00:22:57

```

Seb0815, 2023-01-21:
the code is already merged into main:

in BWifi.cpp 'initWifi' has setAutoReconnect defined and 'handleWebserver' has the needed code
`if (WiFi.status() != WL_CONNECTED) {
    Serial.println(F("WiFi is disconnected, try to reconnect..."));
    WiFi.disconnect();
    WiFi.reconnect();
    AddtoMsgView(String(millis()) + ": WLAN ERROR! try to reconnect");
    delay(1000);
  }
`
I haven't tested the current FW version yet, I'm still using my code - but the changes are already integrated

Seb0815, 2023-01-22:
the initial connect should be handled by WiFiManager, maybe the version of the lib your are using is not the best way to support (I'm running 2.015-rc1)?

giovanne123, 2023-01-23:
I'm using the same lib-version:

```
[env]
lib_deps =
    https://github.com/tzapu/WiFiManager/archive/refs/tags/v2.0.15-rc.1.zip
...
```

```
Scanning dependencies...
Dependency Graph
|-- WiFiManager @ 2.0.15-rc.1
|   |-- DNSServer @ 2.0.0
...
```

giovanne123, 2023-01-25:
No further investigation, seems only I have the problem and therefore I will use the workaround to limit the AP timeout until reboot.

closed 2023-01-25 by giovanne123