mariolukas/Bluetti_ESP32_Bridge #32: [Feature] AC/DC switch status change should update MQTT topic

gerab, 2023-01-26:
It would be nice to be sure about the current status before the change. It's possible to just override it with MQTT, but a feedback from device sounds much better.
Currently there is no update to */command/* at all.

AlexBurghardt, 2023-01-28:
I am not sure if this is possible at all. But I can try something, once I get the spare time needed to do this. Maybe it will work.
I tried to synchronize it with the status via iobroker script but it turned out that it is not easy doing it this way.

gerab, 2023-01-29:
It shouldn't update the 'command' topic but instead send the ac/dc status with other stats. If it's somewhere in dataset from BT it should be possible. I'll check it if you won't do it first.

gerab, 2023-01-29:
I have not tested it yet, but as I see, you made progress very fast, thank you
```
{AC_OUTPUT_ON,              0x00, 0x30, 1, 0, 0, BOOL_FIELD},
  {DC_OUTPUT_ON,              0x00, 0x31, 1, 0, 0, BOOL_FIELD},
```
https://github.com/mariolukas/Bluetti_ESP32_Bridge/pull/33

gerab, 2023-01-29:
I need some help here @mariolukas @AlexBurghardt 
Spent a few hours and found out that it's actually the filter in PayloadParser.cpp:58 that filters 
bluetti/*/state/ac_output_on
bluetti/*/state/dc_output_on

and if I comment out that if block it's work and show updates.

However I'm not able to send messages anymore and device reboots with message:

>publish topic for field: device_type
>Error while publishTopic! 'device_type' can't be empty, reboot device)

AlexBurghardt, 2023-01-30:
Look into PayLoadParser of my Pull Request. It is part of my improvements.

gerab, 2023-01-30:
@AlexBurghardt I'm an idiot, I was looking at your 'main' branch. Thanks, It works now.

closed 2023-01-30 by gerab