mariolukas/Bluetti_ESP32_Bridge #36: [Feature] keep alive messages

gerab, 2023-01-31:
Would be nice to have some kind of keep-alive command. At the moment, bluetti will turn off the main board on ~3-4 hours of inactivity. I'm sending a message to switch DC on/off every hour and it works, but it makes DC outputs unusable.
Original code from warhammerkid contains auto_sleep_mode command. Does it worth to be implemented? Even if it won't work maybe it could keep bluetti alive?

gerab, 2023-02-01:
I missed that it's also implemented in PR, pulled the required changes and it seems to work, testing

gerab, 2023-02-03:
As today was the first day without power outages and the station was primarily untouched physically and programmatically, only a scheduled task of switching the screen timeout occurred, and the station successfully fell asleep 😪, back to DC jerking. Still in search of a way to keep the station powered on indefinitely.

SmokyBob, 2023-02-10:
@gerab from the behaviour you are experiencing seems like EcoMode is on.

My eb3a the board is directly connected to the dc usb output and it's on for days (goes off when I disconnect to upload a new version of the code), but I had to turn off EcoMode

gerab, 2023-02-10:
> seems like EcoMode is o

nope, https://community.bluettipower.com/t/does-ac200max-auto-power-off/7807

SmokyBob, 2023-02-10:
ah ok, from the forum post seems that ecomode is still "applied" if nothing is connected / using power.
That explains why it's staying on in my case, as I have the ESP32 board connected and using DC power.

gerab, 2023-02-10:
There are two different things: "eco mode," which I believe will only switch off the inverter, and the other thing, which doesn't have a name or menu options in settings that will turn off the station entirely if idle. And yes, it doesn't bother you because you use DC constantly. And my station is just waiting for a command to put AC on while entirely idle.

gerab, 2023-02-20:
It seems like there is no such option for AC200M, at least with the latest global firmware.
I will close until that might be changed.

closed 2023-02-20 by gerab