mariolukas/Bluetti_ESP32_Bridge #37: Output connect status on RGB LED

viper4gh, 2023-02-01:
I have a M5Stack Atom Lite, which has an addressable RGB LED (SK6812) built in.
Because of that I added some code based on the FastLED Library, which shows some connection states on it.
I added the complete code conditionally, that means as long as USE_FASTLED is set to false in the config.h you don't need the FastLED library in the lib dependencies.
The logic behind: On startup of the device you see then the colors for successful connection of WiFi, MQTT, Bluetooth and sending data to the MQTT server.
The color codes and the GPIO PIN are defined in the config.h.
Further if all is working and one of these colors will be active again, it shows a connection problem with the connection which is assigned to the color.
If it is ok I could do a Pull Request.

matthewhadley, 2023-03-10:
As someone also using M5Stack Atom Lites for this purpose, I'd be interested to see the PR

viper4gh, 2023-03-10:
@matthewhadley
As long as the PR is not requested, you can use my [Fork](https://github.com/viper4gh/Bluetti_ESP32_Bridge_M5StackAtomLite). 
My fork is in sync with the main repository and only the LED feature is additionally. 
Simply enable the FastLED lib dependency in the platformio.ini, set USE_FASTLED to true in config.h and maybe change the BLUETTI_TYPE in config.h.

matthewhadley, 2023-03-10:
Thanks @viper4gh, I will check it out.

I have [my own fork](https://github.com/matthewhadley/Bluetti_ESP32_Bridge) with some additions:
- define device name using a build flag
- separate MQTT topics for device status vs a json string