mariolukas/Bluetti_ESP32_Bridge #39: change grid charging speed?

Seb0815, 2023-02-01:
With fragments of #33 I was able to get lot more insights from my EP500P, thanks to Alex.

The EP500P is able to change the grid charging speed, any idea to get the necessary page/offset to read/write this information?
It's only possible direct on the device, not via mobile app.

AlexBurghardt, 2023-02-01:
Well, I tried to modify (extend) the page 0x0B request which did not work.
My idea was to access the parameters that can be set via HMI, too.
I only was able to implement the autosleep setting because I did not see other plausible changes in the 0x0B page when I changed other settings via HMI. Maybe these settings can not be changed via BT or I did not look at it long enough.
I spent not very much time on this topic, to be honest. Since you can not change it via app it might be not implemented (yet), that's my guess.

warhammerkid, 2023-02-03:
I have spent quite a bit of time looking for it and was unable to find it. I’m certain there’s no way to read the value from page 0x00 or 0x0B (nothing changes when that setting is changed), so if it’s writable somewhere on page 0x0B the only way to discover it would be to write values to every potential offset and hope you don’t brick your device in the process. I’d have to pull up my notes, but I think pages 0x03 and 0x04 spit out a bunch of garbage, so I assumed (but never verified) they’re part of the firmware update process. Page 0x13 is for wifi config. There’s also 6 or 8 bytes on some other page, but they don’t have anything to do with the grid current setting either.

AlexBurghardt, 2023-02-03:
@warhammerkid: Stephen, thank you very much for your excellent reverse engineering. I checked your python project and did not find the temperature values that are displayed on the HMI's secret BMS screen. I assume, that you did not find these values on page 0x00. Am I right?

warhammerkid, 2023-02-05:
> I assume, that you did not find these values on page 0x00. Am I right?

@AlexBurghardt I think 0x5F is probably a temperature sensor of some type, but I haven't spent any time figuring out how it works. I noticed that the value was lower in the logs I took in the winter (when my AC300 was outside) and higher in logs I took when it was inside. However, I haven't spent any time figuring out how to decode it.

AlexBurghardt, 2023-02-11:
@warhammerkid: I tried that offset but I got a value that obviously had nothing to do with the displayed temperature values. I looked if any byte would read a  corresponding value, but nothing seemed to be fitting. Well, I just spent just another two hours, additionally I also requested a second "page portion" above 7F and found nothing but zero values. Maybe someone else has more luck.