mariolukas/Bluetti_ESP32_Bridge #40: Trying to get it running for EP600

gedongi, 2023-02-02:
Hi everyone,

i am trying since days to get it run for the EP600.
Everything is running so far...
BLE connection
[image]

MQTT server
[image]

The problem facing is that there are no states creating in IO broker objects, commands are created but no values
[image]

Can anyone help me to figure out the problem or at least check if the ESP_32_Bridge is able to get the datas from the EP600 as the other devices?

Regards

gerab, 2023-02-02:
EP600 is not supported yet, as I know. 
https://github.com/mariolukas/Bluetti_ESP32_Bridge/blob/main/Bluetti_ESP32/PowerStation.h

But You can try to test it and show us a confirmation of work.

try my Frankenstein from here: 
shorturl.at/cFGR9

gedongi, 2023-02-02:
first of all, thank you for fast reply and first attempt for solving...
with your sketch it works as before...
Connection established but no data to the MQTT server nor any state objects inside MQTT...

19:57:58.056 -> *wm:AutoConnect: SUCCESS 
19:57:58.056 -> *wm:STA IP Address: 192.168.178.100
19:57:58.056 -> 
19:57:58.056 -> IP address: 
19:57:58.056 -> 192.168.178.100
19:57:58.056 -> MDNS responder started
19:57:58.056 -> HTTP server started
19:57:59.777 -> BLE Advertised Device found: Name: EP60022xxxxxxx, Address: 94:b5:55:93:bf:01, manufacturer data: 424c424c5545545449, serviceUUID: 0000ff00-0000-1000-8000-00805f9b34fb, rssi: -83
19:57:59.823 -> Connecting to MQTT at: 192.168.178.77:1885
19:57:59.855 -> Connected to MQTT Server... 
19:57:59.855 -> subscribe to topic: ac_output_on
19:57:59.855 -> subscribe to topic: dc_output_on
19:57:59.855 -> Forming a connection to 94:b5:55:93:bf:01
19:57:59.855 ->  - Created client
19:58:00.210 ->  - Connected to server
19:58:00.895 ->  - Found our service
19:58:00.895 ->  - Found our Write characteristic
19:58:00.895 ->  - Found our Notifyite characteristic
19:58:00.941 -> The characteristic value was: 
19:58:01.051 -> activate relais contact
19:58:01.051 -> We are now connected to the Bluetti BLE Server.
19:58:01.051 -> Write Request FF02 - Value:  0103 000a 0028 65d6
19:58:01.082 -> F01 - Write Response
19:58:01.082 -> 0183 0301 31
19:58:01.082 -> MQTT Message arrived on topic: bluetti/EP60022xxxxxxx/command/ac_output_on Payload: 
19:58:01.082 -> Write Request FF02 - Value:  0106 0bbf 0000 ba0a
19:58:01.175 -> F01 - Write Response
19:58:01.175 -> 0186 02c3 a1
19:58:01.668 -> MQTT Message arrived on topic: bluetti/EP60022xxxxxxx/command/dc_output_on Payload: 
19:58:01.668 -> Write Request FF02 - Value:  0106 0bc0 0000 8bd2
19:58:01.701 -> F01 - Write Response
19:58:01.701 -> 0186 02c3 a1

at MQTT IO Broker
[image]

I don't understand why the states are not created. 
any other suggestion?

warhammerkid, 2023-02-03:
Well, that’s kind of a good sign? Bluetti has been using the MODBUS protocol for their other devices, and the debug logs suggest that they’re still using it for the EP600. However, a response starting with “018#” is how you’re supposed to indicate that there’s an error processing the request (you add 0x80 to the request type, and then the follow that with the error code to describe the kind of error). Presumably all the data for the EP600 will need to be reverse engineered from scratch.

The easy way to do this involves using the bluetooth debug tools for iOS or Android and collecting bluetooth logs while you do stuff. You’ll want to keep track of the time at which you do things so that the actions and logs can be synced up later. From there that should give enough info to build a tool to dump all the data exposed over bluetooth, and then the fun begins in trying to identify what it all means.

gedongi, 2023-02-03:
Stephen, thanks for commenting. I attached a txt file sniffed from the nRF connect app. Could you explain to me how i need to interprate the figure streams after "Notification received from 0000ff01-0000-1000-8000-00805f9b34fb, value:".
For the beginning i'am highly interested to get the battery status percentage, the actual power from my PV modules and the total produced power from the PV. How i can get this to the MQTT server?
Could you short analyze the txt file and it contains what you expected?
I appreciate any kind of help to get it run with the EP600.

[Log 2023-02-03 16_11_42.txt](https://github.com/mariolukas/Bluetti_ESP32_Bridge/files/10580298/Log.2023-02-03.16_11_42.txt)

warhammerkid, 2023-02-05:
@gedongi This definitely looks like what I'd expect but it's missing a crucial piece - the bluetooth communications from your phone to your EP600. Bluetti uses a protocol called [MODBUS](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) to communicate, and with this protocol the response (from your EP600) isn't sufficient on its own to figure out how to communicate with it. The end goal is to build a table [like this](https://doc.chromedshark.com/bluetti/ac300.html) of the various addresses, what they contain, and how to parse them. However, the responses only include the data, not the locations where that data can be found, so we need the requests to start the process of reverse engineering things.

If the data structure is similar to their other power stations, we can start with a bunch of stuff already reverse engineered. If not, then the process is similar to what you just went through. If it's a measurement, we look for that specific measurement in the data. If it's a switch or toggle or something, then we look for what request the mobile app sends to the EP600 to change that setting. Once everything used by the app is hooked up, then we can also do some exploration to see if there are other things we can extract (like the cell voltages for an attached battery pack).

warhammerkid, 2023-02-06:
Took another look at the data you provided. One of the certain things I can look for is the "EP600" text, and we can find it... but it's not in the form it is for other devices. Instead, it shows up in the data as `PE060\x00`, with every other character swapped. It shows up where I'd expect it, but again if I assume that all the other data is where I'd expect it I get a bunch of stuff that doesn't make sense. All this is to say that it's likely going to take some effort to support.

warhammerkid, 2023-02-07:
My C++ skills are pretty poor, and I haven't spent much time looking at the code here, but assuming I'm reading that screenshot correctly, then a device header like this might work to get the battery percent into your MQTT server:

```cpp
static device_field_data_t bluetti_device_state[] = {
  {TOTAL_BATTERY_PERCENT, 0x00, 0x66, 1,0,0, UINT_FIELD},
};

static device_field_data_t bluetti_device_command[] = {};

static device_field_data_t bluetti_polling_command[] = {
  {FIELD_UNDEFINED, 0x00, 0x66, 0x02 ,0 , 0, TYPE_UNDEFINED}
};
```

gedongi, 2023-02-11:
So, finally i got everything i wanted so far to my MQTT server. Here my values i am transfering...

[image]

There was a small code issue on the PayloadParser.cpp which i changed to get it run.
Thank you Guys for your support...
Next step will be...switching the exchange the ESP32 board to my RPI with the @warhammerkid  bluetti_mqtt code. Probaply i need your support again...
cheers...

closed 2023-02-11 by gedongi

Patrick762, 2023-07-29:
PR for the EP600 in this Repository is #68 

@gedongi can you share what values you got working?

gedongi, 2023-07-29:
@Patrick762 i got it running as follow
total battery in %
total power in KW from the PV modules
output battery to consumers in W
total generated power in kWh (missing on screen dump)
here under check screen dump from my visu...the green frames are the values from the EP600.
[image]

I saw your PR#68. You found some new values from the SOC, these are only readable or also writebale?
Can you share the ModBus addresses for the SOC low/high?
You study in Reutlingen? I live near HN...😊

Patrick762, 2023-07-29:
SoC low is at `0xE6` and SoC high at `0xE7`.
I hope I find some more values (after my current clamps are swapped and facing the right direction... ).

Here's the file where I got some of the values from:
https://github.com/warhammerkid/bluetti_mqtt/blob/main/bluetti_mqtt/core/devices/ep600.py

gedongi, 2023-07-29:
could you share also the format as for battery percentage like...

{TOTAL_BATTERY_PERCENT, 0x00, 0x66, 1,0,0, UINT_FIELD},

what is the format for that fields?
{SOC_LOW, 0x00, 0xE6, 1,0,0, UINT_FIELD},
{SOC_HIGH, 0x00, 0xE7, 1,0,0, UINT_FIELD}

decimal, UINT?

Patrick762, 2023-07-29:
should be uint (percentage)

gedongi, 2023-07-29:
puhhh...the response from EP600 to the E6/E7 address is not succesful...

20:56:15.285 -> Write Request FF02 - Value:  0103 00e6 0001 65fd
20:56:15.329 -> F01 - Write Response - Value:0183 0301 31
20:56:25.311 -> Write Request FF02 - Value:  0103 00e7 0001 343d
20:56:25.350 -> F01 - Write Response - Value:0183 0301 31

will dive deeper in again...

mariolukas, 2023-08-16:
> PR for the EP600 in this Repository is #68
> 
> @gedongi can you share what values you got working?

is merged now.