mariolukas/Bluetti_ESP32_Bridge #47: MQTT errors for device_status since last updates

viper4gh, 2023-02-14:
Hi,
I use iobroker with MQTT Adapter as MQTT Broker with an AC200Max.
Since the last Updates the iobroker log shows the following error one time per minute:
Client [Bluetti_ESP32] Cannot parse "bluetti/AC200M*************/state/device_status": {"MQTTconnected":1", "BTconnected":1}

With the version from January 21st I don't get these errors.

Further with the older version I used AC200M-... as Bluetti Devide ID, now it only works without "-". Maybe that was my fault and it was it without the "-" all the time and the ID string is compared more restrictive.
That is no big problem, it is only a complete other device in iobroker now, because the datapoint folder is the device ID.

AlexBurghardt, 2023-02-14:
I am using this adapter as a client with mosquitto as mqtt server. This works without such error messages. It seems as if the adapter has a problem with json formatted payload.

viper4gh, 2023-02-15:
But was it not already the json payload on the version  from January 21st, because there it was working without errors.
The problem exists since I add the updates from the recent days.

mariolukas, 2023-02-15:
I do also prefer the influx data format for my logging system. I will add a feature which makes the data format selectable by  a config value.

viper4gh, 2023-02-15:
I checked it again, the log shows an error, but the data is still used.
And the old version vom January 21st seems not to send this info in each interval, I checked it in the MQTT log in debug mode. This is the cause why the problem was not there before.

But is maybe a simple typo, look at the error message again:
{"MQTTconnected":1", "BTconnected":1}
The quotation marks. There is one quotation mark before the comma after the first 1, which should not be there.

viper4gh, 2023-02-15:
https://github.com/mariolukas/Bluetti_ESP32_Bridge/blob/main/Bluetti_ESP32/MQTT.cpp#L277
`String value = "{\"MQTTconnected\":" + String(isMQTTconnected()) + "\", \"BTconnected\":" + String(isBTconnected()) + "}"; `
I think it should be:
`String value = "{\"MQTTconnected\":" + String(isMQTTconnected()) + ", \"BTconnected\":" + String(isBTconnected()) + "}"; `

viper4gh, 2023-02-15:
I tried the solution in my fork and it is working without errors now.
The typo described above was the cause.
Now the datapoint is also correctly identified as JSON in iobroker, before it was identified as string.

closed 2023-02-21 by mariolukas