mariolukas/Bluetti_ESP32_Bridge #50: On AC200M commands AC and DC work, but states are shown

SvHeGithub, 2023-02-15:
Hi,
i used todays latest main and tried to get it running together with an iobroker as mqtt server.
The following works:
- Connection to iobroker as mqtt server
- Controlling AC and DC using the commands

What I do not see (and others seem to) are all the states like current power and i.e. total battery percent.
I added some debug code in parse_bluetooth_data() the only page 0x00 offset 0x0A length 7, page 0x00 offset 0x46 length 14 and page 0x0B offset 0xB9 length 14. All with some follow data, which not looked to deep into.
The listest data matches the values in bluetti_polling_command for the AC200M.

Thus is I wonder if I am doing something wrong or missing something, if the bluetti_polling_command needs to extended or if there is any other issue. As other person mention that AC200M is working at them I wonder what is the difference to my project (BLUETTI_TYPE AC200M i have defined in the config header).

Any help is welcome!

Regards,
SvHeGithub

AlexBurghardt, 2023-02-16:
I will do a PR with extended dataset for AC200M ASAP. I guess this will help you.

SvHeGithub, 2023-02-16:
Hi @AlexBurghardt
Sounds great. I will Test the PR as soon as i see it here (and i find the time).
Regards 
SvHe

mariolukas, 2023-02-17:
Merged with #51

SvHeGithub, 2023-02-17:
Hi @AlexBurghardt,

i tested the new main, but still i have problem with my AC200M and i even have new once.

1.
I still do not see any new states, just the once i had before.

2.
I am now regularly running to
Error while publishTopic! 'device_type' can't be empty, reboot device)
Thus is reboots.
To avoid the reboot for now did the following:
  //sometimes we get empty values / wrong vales - all the time device_type is empty
  if (map_field_name(field_name) == "device_type" && value.length() < 3){

    Serial.println(F("Error while publishTopic! 'device_type' can't be empty, (not) reboot device)"));
    **return;
    //ESP.restart();**   
  } 

3.
From parse_bluetooth_data the only time it trys to publish something is for the device_type.
No other valid bluetti_device_state is found in the if check.

I wounder what is the different from mine to your setup.
Any idea?
Using the mobile app (not in parallel), bluetooth is working fine.

Regards,
SvHe

AlexBurghardt, 2023-02-17:
Is your configuration in config.h correct? Is really Device_AC200M.h being included?

SvHeGithub, 2023-02-17:
Hi @AlexBurghardt 
Yes i do (tested it with some compile errors in the file).
Any other idea? Can you post some of your serial Monitor Bluetooth data so i can compare it.
Regards 
SvHe

AlexBurghardt, 2023-02-17:
I can, but it will take some time. By tomorrow latest...

AlexBurghardt, 2023-02-17:
Ok, this is an excerpt from my serial log:
----------------------------------------------------------------------------------------------
21:36:58.209 -> We are now connected to the Bluetti BLE Server.
21:36:58.209 -> Write Request FF02 - Value:  0103 000a 007f 2428
21:36:58.209 -> Update Web events
21:36:58.209 -> 
21:36:58.405 -> F01 - Write Response
21:36:58.405 -> 0103 fe41 4332 3030 4d00 0000 0000 0003 
21:36:58.405 -> f9ab  fc23 4602 0700 0000 0000 001d a300 
21:36:58.405 -> 061c 7b00 068d 4700 0100 0000 0000 0000 
21:36:58.405 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:36:58.405 -> 0000 0000 0000 1500 0000 0000 0000 0000 
21:36:58.405 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:36:58.405 -> 0000 0000 0000 00c0 0000 0000 0000 0000 
21:36:58.405 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:36:58.405 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:36:58.437 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:36:58.437 -> 0000 0000 0000 0314 3800 0000 1500 2800 
21:36:58.437 -> 0100 0114 3800 1500 0000 6401 2c16 bc02 
21:36:58.437 -> 5801 4201 4201 4201 4201 4201 4201 4201 
21:36:58.437 -> 4201 4201 4201 4201 4201 4201 4201 4201 
21:36:58.437 -> 4200 4000 4000 3f00 0000 0000 0000 0000 
21:36:58.437 -> 0000 0000 0000 0000 0000 0000 
21:36:58.437 -> publish topic for field: device_type
21:36:58.437 -> publish topic for field: serial_number
21:36:58.469 -> publish topic for field: arm_version
21:36:58.469 -> publish topic for field: dsp_version
21:36:58.469 -> publish topic for field: dc_input_power
21:36:58.469 -> publish topic for field: ac_input_power
21:36:58.469 -> publish topic for field: ac_output_power
21:36:58.469 -> publish topic for field: dc_output_power
21:36:58.469 -> publish topic for field: power_generation
21:36:58.469 -> publish topic for field: total_battery_percent
21:36:58.469 -> publish topic for field: ac_output_on
21:36:58.501 -> publish topic for field: dc_output_on
21:36:58.501 -> publish topic for field: internal_ac_voltage
21:36:58.501 -> publish topic for field: internal_ac_frequency
21:36:58.501 -> publish topic for field: ac_input_voltage
21:36:58.501 -> publish topic for field: internal_dc_input_voltage
21:36:58.501 -> publish topic for field: pack_max_num
21:36:58.501 -> publish topic for field: internal_pack_voltage
21:36:58.501 -> publish topic for field: internal_cell01_voltage
21:36:58.533 -> publish topic for field: internal_cell02_voltage
21:36:58.533 -> publish topic for field: internal_cell03_voltage
21:36:58.533 -> publish topic for field: internal_cell04_voltage
21:36:58.533 -> publish topic for field: internal_cell05_voltage
21:36:58.533 -> publish topic for field: internal_cell06_voltage
21:36:58.533 -> publish topic for field: internal_cell07_voltage
21:36:58.533 -> publish topic for field: internal_cell08_voltage
21:36:58.565 -> publish topic for field: internal_cell09_voltage
21:36:58.565 -> publish topic for field: internal_cell10_voltage
21:36:58.565 -> publish topic for field: internal_cell11_voltage
21:36:58.565 -> publish topic for field: internal_cell12_voltage
21:36:58.565 -> publish topic for field: internal_cell13_voltage
21:36:58.565 -> publish topic for field: internal_cell14_voltage
21:36:58.565 -> publish topic for field: internal_cell15_voltage
21:36:58.565 -> publish topic for field: internal_cell16_voltage
21:36:58.598 -> F01 - Write Response
21:36:58.598 -> 0000 0000 00de b1
21:37:03.192 -> Write Request FF02 - Value:  0103 0bb9 003f d61b
21:37:03.225 -> Update Web events
21:37:03.225 -> 
21:37:03.368 -> F01 - Write Response
21:37:03.401 -> 0103 7e00 0200 0000 0000 0000 0000 0000 
21:37:03.401 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:37:03.401 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:37:03.401 -> 0000 0000 0000 0000 0000 0000 0000 0017 
21:37:03.401 -> 0211 1526 3900 0000 0000 0000 0000 0000 
21:37:03.401 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:37:03.401 -> 0000 0000 0000 0000 0000 0000 0000 0000 
21:37:03.401 -> 0000 0000 0000 0000 0000 0000 0300 0000 
21:37:03.401 -> 007f 
21:37:03.401 -> publish topic for field: auto_sleep_mode
----------------------------------------------------------------------------------------------
Please note that the checksum of the page 0x00 response is invalid because I edited the serial field for privacy reasons.

AlexBurghardt, 2023-02-17:
I would have to see your code for further support.

SvHeGithub, 2023-02-18:
Hi @AlexBurghardt 
here is my startup sequence.
I removed others doing and I am only polling the states.
I also added a Milli Info to know how much times passed as i do not have the time from my serial monitor.
What i see is that the response is to short, but i get several of them.
If you have any idea, that is welcome. ;-)

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
Loading Values from EEPROM
*wm:AutoConnect 
*wm:Connecting to SAVED AP: Home
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.0.227

IP address: 
192.168.0.227
MDNS responder started
webserver BT/MQTT variable logging disabled...
HTTP server started
BLE Advertised Device found: Name: AC200Mxxxxxxxxxxxxxxxxx, Address: 00:15:83:92:6a:3f, appearance: 960, serviceUUID: 0000ff00-0000-1000-8000-00805f9b34fb, rssi: -80
Connecting to MQTT at: 192.168.0.127:1883
Connected to MQTT Server... 
subscribe to topic: dc_output_on
subscribe to topic: ac_output_on
subscribe to topic: auto_sleep_mode
subscribe to topic: power_off
Forming a connection to 00:15:83:92:6a:3f
 - Created client
 - Connected to server
 - Found our service
 - Found our Write characteristic
 - Found our Notifyite characteristic
We are now connected to the Bluetti BLE Server.
Milli: 3795
Write Request FF02 - Value:  0103 000a 007f 2428
Milli: 4001
F01 - Write Response
0103 fe41 4332 3030 4d00 0000 0000 0003 
f99a 05c9 
Milli: 4007
F01 - Write Response
fd02 0200 0000 0000 001d a300 061c 7b00 
068d 4600 
Milli: 4509
F01 - Write Response
0100 0000 0000 0000 0000 0000 0000 0000 
3900 0000 
Milli: 5011
F01 - Write Response
F01 - Write Response
0000 00c0 0000 0000 0000 0000 0000 0000 
0000 0000 
Milli: 6514
F01 - Write Response
0000 0000 0100 e600 0000 0001 f400 0000 
0000 0000 
Milli: 6796
Write Request FF02 - Value:  0103 000a 007f 2428
Milli: 6796
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
2202 4200 
Milli: 7307
F01 - Write Response
a800 0000 0000 0314 bb00 0600 1700 2800 
0100 0114 
Milli: 7808
F01 - Write Response
bb00 1700 0000 6401 2c16 bc02 5801 4a01 
4a01 4a01 
Milli: 8309
F01 - Write Response
4a01 4a01 4a01 4a01 4a01 4a01 4a01 4a01 
4a01 4a01 
Milli: 8810
F01 - Write Response
4a01 4a01 4a00 4f00 4900 4f00 0000 0000 
0000 0000 
Milli: 9311
F01 - Write Response
0000 0000 0100 e600 0000 0000 0000 0000 
00a3 01
Milli: 9837

SvHeGithub, 2023-02-18:
Hi @AlexBurghardt,
i reduced the amout of data read at one command to a number that seems to fit to what i so far receive.
That works for me, so i receive the data i need (and not caring on the others).
You log helped me to understand it better, so thanks!!
@mariolukas I think we an document it somewhere that reducing the amount of byte that are read might help (in some setups).
Regards,
SvHe

AlexBurghardt, 2023-02-18:
Hi @SvHeGithub,

Warhammerkid (who reverse engineered the BT protocol) mentioned this in his posts. Requesting larger portions increases the risk of getting corrupted responses. The Bluetti device is responding by sending small parts that have to be concatenated in the ESP32. It seems that this did not work for you.

viper4gh, 2023-03-03:
Hi,

the problem with the corrupted data seems to be right.
Since I updated the AC200M.h where more data have to be transmitted I get more and more cryptic datapoints over time and errors in the iobroker log.
Is there anything we can do here?

[image]
[image]

AlexBurghardt, 2023-03-03:
Hi @viper4gh,

I do not believe that this is related to the bluetooth connection. It seems as if there are corrupted/scrambled mqtt topics  transmitted. Yes, definitely not an issue with BT connection, because the topic names come from <String map_field_name(enum field_names f_name){}> in MQTT.cpp.

If there was a BT issue, the topics name would be correct and the values only would be wrong.

When did you reset the ESP32 before this ocurred?
For how long time did it run properly before it showed this issue?
Maybe it is an issue of the mqtt adapter, are you running it as a broker?

Regards,
Alex

viper4gh, 2023-03-03:
@AlexBurghardt
It is difficult to say, it was active several hours I think.
Because the ESP32 reboots automatically on connection problems, it is further difficult to answer. 
I think it would help a lot if we could add the uptime (in seconds for example) and the publish error counter as publish topic to the MQTT Broker. Then I could log these values to Influx DB and could better answer this question. 

I have the MQTT Broker Adapter for iobroker and it runs as Broker, yes. 
Further the corrupted values began since the new values for the AC200M are in. 
With the version 3 weeks ago I had not such values. 
And it seems that these corrupted values are only copies from the new values. I had never a corrupted value for total battery for example. 

Would it help to test with the old AC200M.h file again?

viper4gh, 2023-03-03:
@AlexBurghardt 
Ok, thanks.
Checking the Serial Output could be time consuming, because I don't know when the problem occurs.
But I will try.

AlexBurghardt, 2023-03-03:
In the part

-----------------------------------------------------------------------------------------------------------------------------
 ESPBluettiSettings settings = get_esp32_bluetti_settings();
->  sprintf(publishTopicBuf, "bluetti/%s/state/%s", settings.bluetti_device_id, map_field_name(field_name).c_str() ); 
  lastMQTTMessage = millis();
->  if (!client.publish(publishTopicBuf, value.c_str() )){
    publishErrorCount++;
    AddtoMsgView(String(lastMQTTMessage) + ": publish ERROR! " + map_field_name(field_name) + " -> " + value);
  }
  else{
    AddtoMsgView(String(lastMQTTMessage) + ": " + map_field_name(field_name) + " -> " + value);
  }
-----------------------------------------------------------------------------------------------------------------------------

of <void publishTopic(enum field_names field_name, String value){}> in MQTT.cpp could happen something unexpected.

The first line marked with "->" assembles the topic name and it's path
The second line (marked with "->" publishes the topic along with it's value converted to string

You could print these strings to the serial console (temporaril)y for debugging.

viper4gh, 2023-03-08:
@AlexBurghardt 
A little Update:
I added the serial output like you described and now testing the third day with it, because the recent 2 days the problem occurs not one single time. And the ESP32 was running around 8 hours each day.
For today, the third day, I thought about what is the difference. And the only difference which is not code related, is the position of the ESP32, because I had in connected to my PC for the serial connection.
Then I connected a longer USB cable and positioned the ESP32 there, where it is normally, next to the powerstation, because normally it is connected directly to it.
And after half an hour the problem occured.
First of all the serial output shows no scrambled publishTopicBuf or value, that means the scrumbling must happen later.

And then I thought about the WiFi signal, because I have an AP on the same floor and one AP a floor above.
I have to check it again, but I think the recent two days without the problem the ESP32 was always connected to the AP on the same floor, where the ESP32 shows a RSSI around -60 dbm.
But at the position next to the powerstation the ESP32 sometimes connecting to the AP a floor above, where the signal is worse. I don't know why it is connecting there, but it does it. And there the RSSI is around -85 dbm, which is a critical signal strength.

I will test a bit more.
I will add the RSSI to publishDeviceState, which could help to see if the problem only occurs on bad Wifi Signal.

AlexBurghardt, 2023-03-10:
@viper4gh
If you are browsing the ESPs status webpage all the time: this could be a root cause!
After I extended the AC200M dataset and monitored the output using the status webpage I got TCP/IP/WIFI warnings written to the serial log and strange behaviour. The error messages diappeared once I closed the browser and left the webserver alone.

viper4gh, 2023-03-10:
@AlexBurghardt
Unfortunately I had the problems today again, but without having the webpage open.
It runs around 3 hours without any problems and then there were publish errors all the time and scrambled data again. But all normal values were still updated correctly. 
And the RSSI was around -65 dbm all the time, which means the WiFi signal quality cannot be a cause.
After a reset it works fine again the rest of the day.
The search goes on... 😐

viper4gh, 2023-03-11:
Is there a possibility to read the CPU Load and free RAM, that we can add these to the device state topic?

AlexBurghardt, 2023-03-16:
@viper4gh:
I doubt that there is a built in function. There are some hints to implement such a function. Just do a google search regarding "cpu load" on ESP32. But I think it just will give you a "guess", It won't be precise at all.

github-ccnt-nib, 2023-04-04:
Hi. I just come across this issue myself. Owner of the (second) AC200M and initially very happy about your development of the ESP32 bridge as it would be a much more power-saving solution than the previous RaspberryPi bridge (that actually was quite complete and working for me). 
After some (minor) changes to the ESP32 code, I was able to compile and upload it to my ESP. After several rounds of trial-and-error (due to the EEPROM save which probably overrides the precompiled values (user/pass/id)), I was finally able to connect both to my broker (needs three tokens: user, pass, ID to be valid) and also the Bluetooth connection to the AC200M is established. I see datagrams and the (very basic) state info device_status (1, 1/0), device (IP, MAC, Uptime) and device_type (AC200M) on the foreseen topics, but nothing else.
I can use the command channels to switch AC and DC on and off, but there are actually NO proper state outputs arriving at the broker. I can see a pattern emerging: pData[1] of 0x00 three times, then pData[1] == 0x03 once, but no publish, then the error "Error while publishTopic! 'device_type' can't be empty, reboot device)" appears (the device_type variable is properly set once, but later is empty). I also bridged this error with a "return before reset" to keep the connection alive a bit longer. 
Anyways, I can try to assist in developing this further, if you care to explain a bit about where to best look.

bascht, 2023-09-15:
I'm stuck at the same point now. In case any one of you had any more success with the AC200M - I'm interested and could supply more debug logs if needed. I'm going back to https://github.com/SSMCD/bluetti2mqtt for the mean time, since that worked okay-ish as long as the AC200M is near my home assistant machine. :smile: