mariolukas/Bluetti_ESP32_Bridge #53: Not connected to MQTT in Home Assistant

Dengor1984, 2023-02-20:
ESP working, but no device BLUETTI in MQTT Home Assistant
[image]
[image]

Dengor1984, 2023-02-20:
[image]

Dengor1984, 2023-02-20:
[image]

mariolukas, 2023-02-21:
The first image shows that the ESP32 is not connected to the MQTT server. The second shows that it is connected to the Server. 

Is this issue solved now? Or can you give a more detailed description of the current state?

Dengor1984, 2023-02-21:
Issue is solved by my self - i have add all sensors by hand in mqtt.yaml and it's works.
But automatically it doesn't add AC200 with ALL sensors and  switches to mqtt!(

Dengor1984, 2023-02-21:
Ps
ESP was connected to mqtt broker and send the topics, but mqtt don't respond it and as result no device in mqtt

Dengor1984, 2023-02-21:
[image]

Dengor1984, 2023-02-21:
[image]

Dengor1984, 2023-02-21:
Switch doesn't worked yet - maybe it with mistakes

closed 2023-02-21 by Dengor1984