mariolukas/Bluetti_ESP32_Bridge #55: IOT component firmware update to version 9014.12 = bluetooth connection is extrem unstable

alterHase77, 2023-03-31:
I updated my AC300 from 9014.10 to 9014.12 because the bluetooth connection was not that stable. Using the newest version of the Bluetti_ESP32_Bridge was ok but i was hoping the firmwareupdate at the bluetti would made it perfect :-)
After the update the bluetooth connection is extrem unstable, near  to unusable.

alterHase77, 2023-04-15:
Is there anyone with the firmware 9014.12 and a stable bluetooth connection? 
As soon as my ESP32S3 connects via bluetooth the connection is interrupted after some seconds or at best after some minutes.
Before i updated my Bluetti to 9014.12 the connection was stable for hours or even for a day or two.
It seems to me like my ESP is overloaded and restarts because of this. Any ideas to solve this issue?

warhammerkid, 2023-04-15:
I've never had a version of the AC300 firmware that didn't occasionally time out or get stuck with Bluetooth, and v9014.12 certainly hasn't seemed any different. Looking at my logs (I'm using `bluetti-mqtt` on a Raspberry Pi), I can see several periods per day where it automatically disconnects and waits to reconnect because it keeps getting back bad data. With `bluetti-mqtt` I have limited control over the Bluetooth stack - I'd always hoped once I had the time to mess around with the ESP32 that I could get it to be smarter about reconnect.

StefanoStream, 2024-05-21:
I have the same issue of @alterHase77 on my dad's AC300.
On my AC300 instead I have a complete different IOT version which is v9041.03 ant the Bluetooth is super stable and I don't have any problems with the ESP32 Wroom for weeks!
I don't know why the IOT version appears so different even though both AC300 has been updated.
Anyway on my dad's ESP32/Bluetti, I'll try to add an ESP32 reset after a few hours. Let me know guys if you found a different workaround.

Spettacolo83, 2024-05-21:
I have the same issue of @alterHase77 on my dad's AC300.
On my AC300 instead I have a complete different IOT version which is v9041.03 ant the Bluetooth is super stable and I don't have any problems with the ESP32 Wroom for weeks!
I don't know why the IOT version appears so different even though both AC300 has been updated.
Anyway on my dad's ESP32/Bluetti, I'll try to add an ESP32 reset after a few hours. Let me know guys if you found a different workaround.

alterHase77, 2024-05-21:
I switched from the esp32 to a raspberry pi and the problem is gone.

StefanoStream, 2024-05-22:
Thank you @alterHase77 so you are not using the ESP32 anymore?
Are you using an Home Assistant plugin that uses the raspberry pi bluetooth directly? If yes, which one?
Thank you

alterHase77, 2024-05-24:
@Spettacolo83 No.
I use the Bluetti to MQTT addon

Spettacolo83, 2024-05-24:
Thank you @alterHase77 do you have a link of that, please?

alterHase77, 2024-05-24:
@Spettacolo83 
https://github.com/SSMCD/bluetti2mqtt