mariolukas/Bluetti_ESP32_Bridge #60: Compilation error: display.h

Ricox1975, 2023-05-18:
When compiling via arduino version: 2.1.0 the following error occurs:

`Users/rxxxxxxx/Documents/Bluetti_ESP32_Bridge-main-lily/Bluetti_ESP32/display.cpp:6:10: fatal error: display.h: No such file or directory
 #include <display.h>
          ^~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: display.h: No such file or directory`

mariolukas, 2023-05-28:
any update on this issue?

sibelle-labs, 2023-06-13:
The same problem with me too ...

When compiling via arduino version: 2.1.0 installed today.
C:\Bluetti_ESP32_Bridge-main\Bluetti_ESP32\display.cpp:5:10: fatal error: display.h: No such file or directory
 #include <display.h>
          ^~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: display.h: No such file or directory

mariolukas, 2023-06-15:
Can you try to change 

`#include <display.h>`

to 

`#include "display.h"`

in display.cpp

jowo19, 2023-06-15:
Thanks for the hint. After I have changed from <display.h> to "display.h" in display.cpp I get:

C:\Users\xxxx\OneDrive\Dokumente\Arduino\Sketches\Bluetti_ESP32_Bridge-main\Bluetti_ESP32\display.cpp: In function 'void initDisplay()':
display.cpp:51:16: error: 'DISPLAY_SDA_PORT' was not declared in this scope
     Wire.begin(DISPLAY_SDA_PORT, DISPLAY_SCL_PORT);
                ^~~~~~~~~~~~~~~~
C:\Users\xxxx\OneDrive\Dokumente\Arduino\Sketches\Bluetti_ESP32_Bridge-main\Bluetti_ESP32\display.cpp:51:16: note: suggested alternative: 'DISPLAY_H'
     Wire.begin(DISPLAY_SDA_PORT, DISPLAY_SCL_PORT);
                ^~~~~~~~~~~~~~~~
                DISPLAY_H
display.cpp:51:34: error: 'DISPLAY_SCL_PORT' was not declared in this scope
     Wire.begin(DISPLAY_SDA_PORT, DISPLAY_SCL_PORT);
                                  ^~~~~~~~~~~~~~~~
C:\Users\xxxx\OneDrive\Dokumente\Arduino\Sketches\Bluetti_ESP32_Bridge-main\Bluetti_ESP32\display.cpp:51:34: note: suggested alternative: 'DISPLAY_H'
     Wire.begin(DISPLAY_SDA_PORT, DISPLAY_SCL_PORT);
                                  ^~~~~~~~~~~~~~~~
                                  DISPLAY_H
exit status 1
'DISPLAY_SDA_PORT' was not declared in this scope

----------
config.h starts with:

#ifndef CONFIG_H
#define CONFIG_H
#include "Arduino.h"

#define DEBUG                 1
// Display config section, comment DISPLAYSSD1306 to disable display
//#define DEBUGDISP 1
//#define DISPLAYSSD1306  1
#define DISPLAY_SCL_PORT 4
#define DISPLAY_SDA_PORT 5
//Uncomment to toggle display reset on start, required for displays like LoRa TTGO v1.0
//#define DISPLAY_RST_PORT 16

//#define RESET_WIFI_SETTINGS   

#define EEPROM_SALT 13374

#define DEVICE_NAME "BLUETTI-MQTT"
#define BLUETTI_TYPE AC200M
...

kgooris, 2023-06-15:
ok, you can do the same for config.h in the file display.cpp file. So change from <config.h> to "config.h"
I will create a pull request including these changes. I made display using platform.io, and this did not compain about it ;)

closed 2023-06-16 by mariolukas