mariolukas/Bluetti_ESP32_Bridge #66: Implent Battery Packs AC200Max?

Hankersern, 2023-06-21:
Hi There,

thank you so much for publishing this project. 
Its my first time I flashed an ESP32 myself, and now I am really happy working with it.

One Thing I am missing: I cannot find any info about the Battery Packs attached to my AC200Max.
In https://github.com/warhammerkid/bluetti_mqtt/pull/45/commits/c4b87323918920e6ea3eec24d81afd30d8d78be2 it looks to me like they added them. 

Could you implent this? 
Would be great!

Greetings from Hamburg

Ricox1975, 2023-07-22:
push

Hankersern, 2023-09-22:
Hey @mariolukas 

dont want to bother you!
Do you have plans to implent this? 

As I have no idea how github really works, could I ask some IT-Guy/Girl to implent this / contribute? 2 Kisten Jever are waiting for the person who does it ;-)

Thanks & 
best regards
Timm

mariolukas, 2023-09-23:
What Battery information do you need?
There was a PR in February where information was extended: 

https://github.com/mariolukas/Bluetti_ESP32_Bridge/pull/51/commits/69f349c0b20f46fbb7ed819f2d32705a5d372d06

all possible values form the bluetti_mqtt Repo you mentioned are included: 
https://github.com/warhammerkid/bluetti_mqtt/blob/main/bluetti_mqtt/core/devices/ac200m.py

Hankersern, 2023-09-26:
I need the Status of the extension packs (% of Load)
On the display of the ac200max its available but I dont have it in the mqtt messages.

dbergens, 2023-10-02:
The values of the individual battery packs can be queried by a command. In the original file, this is queried at this point: 

# Controls
        self.struct.add_uint_field('pack_num', 3006)

Hankersern, 2023-10-05:
> The values of the individual battery packs can be queried by a command. In the original file, this is queried at this point:
> 
> # Controls
> ```
>     self.struct.add_uint_field('pack_num', 3006)
> ```

Hi,

thanks a lot for your help. 
Sadly I have no idea where to use that code. Can you give me a hint?

dbergens, 2023-10-18:
In the original code from warhammerkid you can find the following command section:

  # Controls
        self.struct.add_uint_field('pack_num', 3006)
        self.struct.add_bool_field('ac_output_on', 3007)
        self.struct.add_bool_field('dc_output_on', 3008)
        # 3031-3033 is the current device time & date without a timezone
        self.struct.add_bool_field('power_off', 3060)
        self.struct.add_enum_field('auto_sleep_mode', 3061, AutoSleepMode)

To get the individual battery packs (up to 3 maximum) displayed, it is obviously necessary to send a command to the central unit (1st battery pack). Then the battery pack information is displayed according to the selection.

Using the original from warhammerkid and installing it on a Raspberry Pi, I was able to successfully test this on an AC300.

Joern-HAP, 2024-01-20:
using the ESP code here, the AC200M config file is containing this part:

  //Page 0x00 Battery Details
  //constant value, number off possible battery packs, 3 on AC200M, one internal and two external
  {PACK_NUM_MAX,              0x00, 0x5B, 1, 0, 0, UINT_FIELD },
  
Does I have to change the size = 1 to then 2 for internal + one external battery only, or do I also have to create an additional table for external batteries.  e.g. definition in enum field_names does not contain external batteries.

So it not simple config of 1,2,3, but a extension of todays code required or?