mariolukas/Bluetti_ESP32_Bridge #70: MQTT publish errors & reconnect with no further error messages

bascht, 2023-09-11:
Happening on latest `HEAD` aecaeb064625ef4b8b99db6ff815a73347574653 - with an ESP WROOM 32 (NODEMCU DevKit) . Bluetooth connection to the AC200MAX seems to work (see right hand side), but I cannot get MQTT to work:

[image]

```
p>192236: publish ERROR! total_battery_percent -> 300
195063:skip unknow request! page: 7 offset: d0

198252:skip unknow request! page: 8 offset: 0

0: publish ERROR! total_battery_percent -> 300

204067:skip unknow request! page: 7 offset: d0

207253:skip unknow request! page: 8 offset: 0

210237: publish ERROR! total_battery_percent -> 300

213069:skip unknow request! page: 7 offset: d0

216254:skip unknow request! page: 8 offset: 0

219238: publish ERROR! total_battery_percent -> 300

222075:skip unknow request! page: 7 offset: d0

225254:skip unknow request! page: 8 offset: 0

228250: publish ERROR! total_battery_percent -> 300

231075:skip unknow request! page: 7 offset: d0

234289:skip unknow request! page: 8 offset: 0

237265: publish ERROR! total_battery_percent -> 300

240081:skip unknow request! page: 7 offset: d0

243290:skip unknow request! page: 8 offset: 0

246273: MQTT connection lost, try reconnect
```

Is there any endpoint / method to get more log information or a stack trace of what could be going wrong? I tried connecting to the MQTT host name or IP address verified the MQTT credentials as well as the port and everything seems to be correct.

What could I be missing? Let me know if I can supply more info.

mariolukas, 2023-09-12:
Hi, 
can you try it with https://github.com/mariolukas/Bluetti_ESP32_Bridge/commit/4c1672b991eaf8f7e3bb9ee038a2cffbec173f6e

bascht, 2023-09-12:
@mariolukas thanks for the hints! I tried with 4c1672b (had to cherry-pick aa35fd23af196df236f194278d81de40a744911d, otherwise this wouldn't build) but ran into the same error. 

Do you have any hints how I could get usable debug output? Got a serial connection working but I seem to be using the wrong formatter / output format:

[image]

bascht, 2023-09-13:
Update. I must've hit a weird glitch with the ESP that caused it to pick the wrong baud rate.
At any rate – here's the proper debug log:

```
BLE Advertised Device found: Name: AC200MXXXXXXXX, Address: XX:XX:XX:XX:XX:XX, appearance: 960, serviceUUID: 0xff00
Connecting to MQTT at: <my-mqtt-domain>:1883
Forming a connection to XX:XX:XX:XX:XX:XX
 - Created client
 - Connected to server
 - Found our service
 - Found our Write characteristic
 - Found our Notifyite characteristic
activate relais contact
We are now connected to the Bluetti BLE Server.
Write Request FF02 - Value:  0103 0064 003e 85c5
F01 - Write Response
0103 7c00 0100 6401 2c16 bc02 5801 4501 
4501 4401 
publish topic for field: total_battery_percent
F01 - Write Response
4501 4401 4401 4401 4401 4401 4501 4401 
4501 4401 
F01 - Write Response
4401 4501 4400 5800 5100 4700 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0200 e500 0800 b900 5400 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 001d b0
Write Request FF02 - Value:  0103 07d0 0030 4553
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0800 0029 8674
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0064 003e 85c5
F01 - Write Response
0103 7c00 0100 6401 2c16 bc02 5801 4401 
4401 4401 
publish topic for field: total_battery_percent
F01 - Write Response
4401 4401 4401 4401 4401 4401 4401 4401 
4401 4401 
F01 - Write Response
4401 4401 4400 5800 5100 4700 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0200 e600 0800 b900 5400 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0075 d4
Write Request FF02 - Value:  0103 07d0 0030 4553
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0800 0029 8674
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0064 003e 85c5
F01 - Write Response
0103 7c00 0100 6401 2c16 bc02 5801 4401 
4401 4401 
publish topic for field: total_battery_percent
F01 - Write Response
4401 4401 4401 4401 4401 4401 4401 4401 
4401 4401 
F01 - Write Response
4401 4401 4400 5800 5100 4800 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0200 e600 0800 b900 5400 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 00bf d8
Write Request FF02 - Value:  0103 07d0 0030 4553
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0800 0029 8674
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0064 003e 85c5
F01 - Write Response
0103 7c00 0100 6401 2c16 bc02 5801 4401 
4401 4401 
publish topic for field: total_battery_percent
F01 - Write Response
4401 4401 4401 4401 4401 4401 4401 4401 
4401 4401 
F01 - Write Response
4401 4401 4400 5900 5100 4800 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0200 e600 0800 b900 5400 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 00fa 30
Write Request FF02 - Value:  0103 07d0 0030 4553
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0800 0029 8674
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0064 003e 85c5
F01 - Write Response
0103 7c00 0100 6401 2c16 bc02 5801 4501 
4501 4401 
publish topic for field: total_battery_percent
F01 - Write Response
4501 4401 4401 4501 4401 4401 4401 4401 
4401 4401 
F01 - Write Response
4401 4401 4400 5900 5100 4800 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0200 e500 0800 ba00 5400 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0000 0000 0000 0000 0000 
0000 0000 
F01 - Write Response
0000 0000 0000 0048 e8
Write Request FF02 - Value:  0103 07d0 0030 4553
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0800 0029 8674
F01 - Write Response
0183 02c0 f1
Write Request FF02 - Value:  0103 0064 003e 85c5
F01 - Write Response
0103 7c00 0100 6401 2c16 bc02 5801 4501 
4501 4401 
publish topic for field: total_battery_percent
MQTT lost connection, try to reconnect
Connecting to MQTT at: <my-mqtt-domain>F01 - Write Response:
1883
4501 4501 4401 4501 4501 4501 4501 4501 
4501 4501 
F01 - Write Response
4501 4501 4500 5900 5100 4800 0000 0000 
0000 0000 
[…]

(Continues to loop)
```

bascht, 2023-09-15:
Another update - got it to work and the culprit was: :drum: My MQTT password was too long. :joy: Somehow it just aborted without any error or log message. 

I'm now where at the same point where @SvHeGithub and @github-ccnt-nib are in #50. I can see the `state/device_type` message being published to MQTT, but no other state whatsoever. :-/

Closing this.

closed 2023-09-15 by bascht

mariolukas, 2023-09-16:
> Another update - got it to work and the culprit was: 🥁 My MQTT password was too long. 😂 Somehow it just aborted > without any error or log message.

Thank you for the feedback. I think a hint for the password fields would be a great improvement to prevent log debug sessions.