mariolukas/Bluetti_ESP32_Bridge #71: ESP32 crash after few days

Joern-HAP, 2023-10-03:
Hello,

I'm using the Bluetti_ESP32_Bridge with a WROOM32, but it continuously seems to crash after few days.
Is there any change to introduce a restart e.g. after 24 hours to solve this ?
Where is a good place to put the ESP.restart();?

I already increase the delay functions to have less traffic but do not see improvement and serial seem to simple end somewhen...

BR

Spettacolo83, 2024-05-21:
@Joern-HAP did you find a workaround restarting the ESP? Strangely, I have the same issue on my dad's ESP32/Bluetti but I don't have the same issue on mine

henry07-dotcom, 2024-07-02:
Same here with my connection to MQTT, any update? @Joern-HAP or @Spettacolo83