mariolukas/Bluetti_ESP32_Bridge #72: AC500max no data

mdergithub, 2023-10-12:
I uploaded the code to my ESP32 and received no data.  I found a work-around and I thought I'd post it.  I separated the polling request into smaller chunks and it started working!  Thanks so much for your hard work on this, it's amazing!

You need to modify the DEVICE_AC200M and change the polling to:

static device_field_data_t bluetti_polling_command[] = {
  // Status
  // changed to only one page 0 request (a portion of 7F bytes)
 {FIELD_UNDEFINED, 0x00, 0x24, 0x32 ,0 , 0, TYPE_UNDEFINED},
  {FIELD_UNDEFINED, 0x00, 0x47, 0x4A ,0 , 0, TYPE_UNDEFINED},
  {FIELD_UNDEFINED, 0x00, 0x5c, 0x79 ,0 , 0, TYPE_UNDEFINED},

 {FIELD_UNDEFINED, 0x00, 0x30, 0x32 ,0 , 0, TYPE_UNDEFINED},
 // {FIELD_UNDEFINED, 0x0B, 0xF5, 0x07 ,0 , 0, TYPE_UNDEFINED},
  //Pack Polling
  {FIELD_UNDEFINED, 0x00, 0x5B, 0x25 ,0 , 0, TYPE_UNDEFINED}
 // {FIELD_UNDEFINED, 0x00, 0xBB9, 0x3D ,0 , 0, TYPE_UNDEFINED}

I'm not much of a programmer, so I assume there is a more elegant way to do this or separate the polling better.

mdergithub, 2023-10-12:
Also, if we could get the AC200 max to spit out a number for total_generation other than zero, that would be great.

TxKoiBoy, 2023-11-01:
Similar to this, I have an AC300, and I had to really chop down the bytes requested per block in the the final section of the DEVICE_AC300.h in order to get all my essential readings:

static device_field_data_t bluetti_polling_command[] = {
// Split up the original polling command into smaller chunks
// Original:
//  {FIELD_UNDEFINED, 0x00, 0x0A, 0x28 ,0 , 0, TYPE_UNDEFINED},
//  {FIELD_UNDEFINED, 0x00, 0x46, 0x15 ,0 , 0, TYPE_UNDEFINED},
//  {FIELD_UNDEFINED, 0x0B, 0xB9, 0x3D ,0 , 0, TYPE_UNDEFINED}
//  {FIELD_UNDEFINED, 0x0B, 0xDA, 0x01 ,0 , 0, TYPE_UNDEFINED},
//  {FIELD_UNDEFINED, 0x0B, 0xF5, 0x07 ,0 , 0, TYPE_UNDEFINED},
//  {FIELD_UNDEFINED, 0x00, 0x5B, 0x25 ,0 , 0, TYPE_UNDEFINED}

    {FIELD_UNDEFINED, 0x00, 0x0A, 0x07 ,0 , 0, TYPE_UNDEFINED},
    {FIELD_UNDEFINED, 0x00, 0x24, 0x07 ,0 , 0, TYPE_UNDEFINED},
    {FIELD_UNDEFINED, 0x00, 0x2B, 0x07 ,0 , 0, TYPE_UNDEFINED},
    {FIELD_UNDEFINED, 0x00, 0x56, 0x0E ,0 , 0, TYPE_UNDEFINED}

};

My ESP32 is mere inches from the AC300, my WiFi router is a few yards away, and I'm using the free HiveMQ broker, wherever that is, if this makes any difference.

Cheers,

bbkz, 2024-04-13:
Thank you! This helped me to get the AC200M values!

```cpp
static device_field_data_t bluetti_polling_command[] = {
  // Status
  // changed to only one page 0 request (a portion of 7F bytes)
  {FIELD_UNDEFINED, 0x00, 0x24, 0x32 ,0 , 0, TYPE_UNDEFINED},
  {FIELD_UNDEFINED, 0x00, 0x47, 0x4A ,0 , 0, TYPE_UNDEFINED},
  {FIELD_UNDEFINED, 0x00, 0x5c, 0x79 ,0 , 0, TYPE_UNDEFINED},

  {FIELD_UNDEFINED, 0x00, 0x30, 0x32 ,0 , 0, TYPE_UNDEFINED},
  // {FIELD_UNDEFINED, 0x0B, 0xF5, 0x07 ,0 , 0, TYPE_UNDEFINED},
  //Pack Polling
  {FIELD_UNDEFINED, 0x00, 0x5B, 0x25 ,0 , 0, TYPE_UNDEFINED}
  // {FIELD_UNDEFINED, 0x00, 0xBB9, 0x3D ,0 , 0, TYPE_UNDEFINED}
};
```